opensensor/lightNVR #5: Onvif motion detection events.

xromansx, 2025-03-14:
Would be nice to have recording based on onvif motion detection even similar to https://github.com/rostislavzz/motion-onvif-events

gtxaspec, 2025-03-16:
maybe we could use onvif-simple-server on the NVR in concert with lightNVR? just brainstorming

matteius, 2025-03-16:
I've added a detection(s) framework--support for sod models (cnn face detector verified to work) and a basic motion detection model.   You can extend to add a onvif motion detection as well now, will be easier to do with this framework.

matteius, 2025-03-17:
Looks like to do this Ill need onvif option/support first.

xromansx, 2025-03-18:
Yes, this path would be easier, but not mandatory, but in C it would be quite hard for me at least ) I'm trying to make it running on OpenWRT as and app for now no luck maybe will use it as a docker.

matteius, 2025-03-18:
> There is nice project
> https://github.com/BogdanovKirill/OnvifEventsReceiver maybe we can borrow
> some code from it )

I have a branch working on the default issue with more refactoring, but I
had to head out for a bit.  Will try to get things in a better spot soon.

xromansx, 2025-03-18:
Nice, thanks, will monitor the project.

matteius, 2025-03-19:
We have a docker file now and I fixed the segfault issues -- things are in a better spot now to start working on onivf now.

matteius, 2025-03-23:
I added support tonight for ONVIF detection and adding streams this way -- this plus the detection framework I added a week ago kind of lays the ground work for solving this ticket one day.

matteius, 2025-04-14:
We have this now; we can pull the events from the simple_onvif_service.

xromansx, 2025-04-14:
This are great news, I'll try the latest container, any tips how to configure it, supposed it is manually for now yes ?

matteius, 2025-04-14:
[image]

The pre and post detection buffer settings do nothing at this time, and it may not trigger a recording either as its still under development, but you'll see the events on the recordings and in the live stream it will draw a red box around the screen if motion detected, at this time.

closed 2026-02-01 by matteius