opensensor/lightNVR #20: It seems ignore changes in the configuration.

matiasdelellis, 2025-03-20:
Something I noticed a while ago is that I couldn't change the frontend user. It always requires admin:admin

Now trying to change the parameters to debug #19 . It seems that he ignores almost everything. 🤔 
I don't know. If it's just me, ignore it. 😉 
Changing preferences in the front-end almost always fails, and I haven't even tested it yet.

matiasdelellis, 2025-03-20:
It seems to ignore the -c parameter and use the `/var/lib/lightnvr/lightnvr.ini` file intead `/etc/lightnvr/lightnvr.ini`.

At least I can change some parameters there, and when I restart the service, it works. For example, I can disable the authorization with `auth_enabled` , but changing the `username` and `password` is ignored.

matiasdelellis, 2025-03-20:
D'Oh!. 

> password = admin  ; IMPORTANT: Change this default password!

The comment seems to be causing problems. I deleted it, and changing username and password worked. 🤦🏻 
Then I try the rest of the settings. 😬

matteius, 2025-03-22:
I believe these issues are fixed in 7.0.1 now.

matteius, 2025-03-22:
Solved in 0.8.0

closed 2025-03-22 by matteius