opensensor/lightNVR #30: Since the latest changes in API the live preview is public.

matiasdelellis, 2025-03-22:
I'm already using this app on a small NAS and exporting the fronted via lightNVR -> rathole <-> rathole -> ngnix -> Client..

Well, this leads me to create a small Android application so I can see the cameras live. 😅 

If anyone is interested, all you need is this: 😉 
* https://lindevs.com/play-video-file-using-libvlc-on-android

Before at least it required using a digest authorization:
* https://USERNAME:PASSWORD@HOST/api/streaming/STREAM/hls/index.m3u8

But with the latest changes it was made public as:
* https://HOST/hls/STREAM/index.m3u8

... And needless to say, I would like to maintain some kind of authorization. 🙈 :

matteius, 2025-03-22:
Sorry, I spent 20$ trying to get the bottom of this without much success.

closed 2025-03-22 by matteius

matiasdelellis, 2025-03-22:
Hey. Don't apologize.. Again.. it's great what you achieved! 😃 
Add a donation option in the readme. From experience, you do not recover any investment, but it is good to see the support of users. 😉

matiasdelellis, 2025-03-22:
Wow, Thanks again!.

matteius, 2025-03-22:
Haha I enabled my sponsorship link early on.

Latest build fixes a few more issues that is 0.8.7 -- building now.