opensensor/lightNVR #43: Unable to find image 'lightnvr/lightnvr:latest' locally

gamer06, 2025-04-04:
Hello.
I'm having trouble installing lightNVR.
Both through docker and Source, I get the error:

[image]

matteius, 2025-04-04:
Not sure, worked just now:

```
matteius@matteius-Ryzen9:~$ docker pull ghcr.io/opensensor/lightnvr:latest
latest: Pulling from opensensor/lightnvr
6e909acdb790: Already exists 
73acb74f3774: Pull complete 
462802a760ca: Pull complete 
0d58d0ecead5: Pull complete 
8672465bdb28: Pull complete 
ce21a8f3eb7e: Pull complete 
dc9301e2b14e: Pull complete 
4f4fb700ef54: Pull complete 
f58d86bcaa56: Pull complete 
aeeb8b1340b4: Pull complete 
856d8c35750c: Pull complete 
3f9f31d21fda: Pull complete 
e678968c728f: Pull complete 
Digest: sha256:1a35aaf52678f1ac68e084ba995a3dd7b05eaa0c1d44ca68c972869ef4f23629
Status: Downloaded newer image for ghcr.io/opensensor/lightnvr:latest
ghcr.io/opensensor/lightnvr:latest
```

gamer06, 2025-04-05:
It's strange... I did a new installation of Debian 12, but it gives the same error. Even for "./scripts/build.sh" it gave an error.

[image]

nschimme, 2025-04-13:
Try again with the latest instructions, we fixed this in https://github.com/opensensor/lightNVR/pull/53

matteius, 2025-04-13:
@gamer06 Turns out the instructions were wrong -- they have been updated:  https://github.com/opensensor/lightNVR/pull/53/files

closed 2025-04-13 by matteius